wh201906/SerialTest #2: 能否补充以下使用那些依赖包呢？

taotieren, 2022-02-17:
能否补充以下使用那些依赖包呢？
如果能补充全的话
我这边可以帮忙打包到 Arch Linux 那边。

wh201906, 2022-02-17:
非常感谢！
因为我之前没接触过linux上面的打包，所以我不知道怎么看依赖哪些软件包
对V0.1.2版本的程序跑ldd结果如下
```
VirtualBox:~/SerialTest/src/build$ ldd SerialTest 
	linux-vdso.so.1 (0x00007ffd7ffb4000)
	libQt5PrintSupport.so.5 => /lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007fe372c3e000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fe3725a3000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fe371fb8000)
	libQt5SerialPort.so.5 => /lib/x86_64-linux-gnu/libQt5SerialPort.so.5 (0x00007fe371f9d000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fe371a54000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fe371872000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe371721000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fe371706000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe371514000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe3714f1000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fe371469000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007fe371431000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fe371413000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007fe37130e000)
	libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fe3712e1000)
	libicui18n.so.66 => /lib/x86_64-linux-gnu/libicui18n.so.66 (0x00007fe370fe2000)
	libicuuc.so.66 => /lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007fe370dfc000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fe370df6000)
	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007fe370d71000)
	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007fe370d5b000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fe370c32000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fe372e45000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fe370b7a000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fe370b46000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007fe370a85000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007fe370a58000)
	libicudata.so.66 => /lib/x86_64-linux-gnu/libicudata.so.66 (0x00007fe36ef97000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fe36ef24000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fe36ede7000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fe36edbb000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007fe36edb5000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fe36edad000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fe36ed93000)
```
不知道这些信息有帮助吗

wh201906, 2022-02-17:
Qt当中用到的模块如下
```
QT       += core gui serialport printsupport
android {
    QT += bluetooth androidextras
}
```

taotieren, 2022-02-17:
这两个文件也是需要依赖的吗？
qcustomplot.c and qcustomplot.h

taotieren, 2022-02-17:
```bash
 yay -Ss qcustomplot                          
aur/mingw-w64-qcustomplot-qt5 2.1.0-1 (+5 0.04) 
    Plotting library for Qt5 (mingw-w64)
community/qcustomplot-doc 2.1.0-1 (3.5 MiB 10.9 MiB) 
    Qt C++ widget for plotting and data visualization (docs)
community/qcustomplot 2.1.0-1 (469.4 KiB 1.7 MiB) 
    Qt C++ widget for plotting and data visualization

```
Arch 里面有这个包
我添加下依赖应该是没问题的

wh201906, 2022-02-17:
是的，但是目前编译的时候qcustomplot是直接作为源码编译进去的，不是依赖库。不知道这个有影响没

taotieren, 2022-02-17:
不影响，我这边打 补丁也可以

wh201906, 2022-02-17:
好的，辛苦了！

taotieren, 2022-02-17:
```bash
❯ yay -Ql qcustomplot 
qcustomplot /usr/
qcustomplot /usr/include/
qcustomplot /usr/include/qcustomplot.h
qcustomplot /usr/lib/
qcustomplot /usr/lib/libqcustomplot.so
qcustomplot /usr/lib/libqcustomplot.so.2
qcustomplot /usr/lib/libqcustomplot.so.2.1
qcustomplot /usr/lib/libqcustomplot.so.2.1.0
qcustomplot /usr/share/
qcustomplot /usr/share/doc/
qcustomplot /usr/share/doc/qcustomplot/
qcustomplot /usr/share/doc/qcustomplot/changelog.txt

```

taotieren, 2022-02-17:
需要改成调用库的方法
不适合直接调用 源文件
```bash
❯ qmake
WARNING: Failure to find: qcustomplot.cpp
```

taotieren, 2022-02-17:
还没调试完，这是没完成的 PKBUILD
```bash
# Maintainer: taotieren <admin@taotieren.com>

_pkgname=SerialTest
pkgname=serialtest
pkgver=0.1.2
pkgrel=1
pkgdesc="Serial port test tool on Win/Linux/Android, with realtime plotting, shortcut | Qt开发的跨平台串口助手，带实时绘图和快捷发送面板"
arch=('any')
url="https://github.com/wh201906/SerialTest"
license=('GPL2.1')
provides=(${_pkgname})
conflicts=(${pkgname}-git)
#replaces=(${pkgname})
depends=('qcustomplot' 'qt5-serialport')
makedepends=('qt5-tools')
backup=()
options=('!strip')
install=${pkgname}.install
source=("${_pkgname}-${pkgver}.tar.gz::https://ghproxy.com/${url}/archive/refs/tags/V${pkgver}.tar.gz"
        "${pkgname}.install")
sha256sums=('e386cea5b8828b83b9800839e6f72a45304b38300921e6c71490a1e03d607e7e'
            '303f34246c0d341e1093d6e486e2cbfdbbb0d122d751de649f562ebac37777b4')

prepare() {
#     sed -i -e "s|#include "qcustomplot.h"|#include <qcustomplot.h> |g"  "${srcdir}/${_pkgname}-${pkgver}/src/mycustomplot.h"
#     sed -i -e "s|#include "qcustomplot.h"|#include <qcustomplot.h> |g"  "${srcdir}/${_pkgname}-${pkgver}/src/mycustomplot.cpp"
    ln -sf "/usr/include/qcustomplot.h"  "${srcdir}/${_pkgname}-${pkgver}/src/qcustomplot.h"
}

build() {
    cd "${srcdir}/${_pkgname}-${pkgver}/src/"
    qmake
    make
}

package() {
    install -dm0755 "${pkgdir}/usr/bin/" \
                   "${pkgdir}/usr/share/${pkgname}/"

    cd ${srcdir}/${_pkgname}-${pkgver}/src/
#    cp --preserve=mode -r ${_pkgname} language config themes slave  "${pkgdir}/usr/share/${pkgname}/"

#    cd pkg/
#    cp --preserve=mode -r icons "${pkgdir}/usr/share/"

    #ln -sf  "/usr/share/${pkgname}/${_pkgname}" "${pkgdir}/usr/bin/${pkgname}"
#    install -Dm755 "${srcdir}/${pkgname}" "${pkgdir}/usr/bin/${pkgname}"
    install -Dm0644 "${srcdir}/${_pkgname}-${pkgver}/LICENSE" "${pkgdir}/usr/share/licenses/${pkgname}/LICENSE"
#    install -Dm644 "${srcdir}/${_pkgname}-${pkgver}/${_pkgname}/pkg/io.github.skiars.${pkgname}.desktop" "${pkgdir}/usr/share/applications/io.github.skiars.${pkgname}.desktop"
#    install -Dm644 "${srcdir}/${_pkgname}-${pkgver}/${_pkgname}/pkg/20-usb-serial.rules" "${pkgdir}/etc/udev/rules.d/20-usb-serial.rules"
#    install -Dm644 "${srcdir}/${_pkgname}-${pkgver}/${_pkgname}/pkg/io.github.skiars.${pkgname}.metainfo.xml" "${pkgdir}/usr/share/metainfo/io.github.skiars.${pkgname}.metainfo.xml"
}

```

wh201906, 2022-02-17:
@taotieren 我在dev分支下更新了qmake文件，现在应该支持调用库了。你可以
````
git clone https://github.com/wh201906/SerialTest -b dev
````
之后看下能不能成功编译

taotieren, 2022-02-17:
编写开发版的 PKGBUILD

```bash
# Maintainer: taotieren <admin@taotieren.com>

_pkgname=SerialTest
pkgname=serialtest-git
pkgver=V0.1.2.r4.g93928ef
pkgrel=1
pkgdesc="Serial port test tool on Win/Linux/Android, with realtime plotting, shortcut | Qt开发的跨平台串口助手，带实时绘图和快捷发送面板"
arch=('any')
url="https://github.com/wh201906/SerialTest"
license=('GPL2.1')
provides=(${_pkgname})
conflicts=(${pkgname}-git)
#replaces=(${pkgname})
depends=('qcustomplot' 'qt5-serialport')
makedepends=('qt5-tools' 'git')
backup=()
options=('!strip')
install=${pkgname}.install
source=("${_pkgname}::git+${url}.git"
        "${pkgname}.install")
sha256sums=('SKIP'
            '303f34246c0d341e1093d6e486e2cbfdbbb0d122d751de649f562ebac37777b4')

pkgver() {
    cd "${srcdir}/${_pkgname}/"
#     git checkout dev
    git describe --long --tags | sed 's/\([^-]*-g\)/r\1/;s/-/./g'
}

prepare() {
#     sed -i -e "s|#include "qcustomplot.h"|#include <qcustomplot.h> |g"  "${srcdir}/${_pkgname}-${pkgver}/src/mycustomplot.h"
#     sed -i -e "s|#include "qcustomplot.h"|#include <qcustomplot.h> |g"  "${srcdir}/${_pkgname}-${pkgver}/src/mycustomplot.cpp"
#     cd "${srcdir}/${_pkgname}/"
#     git checkout dev
    ln -sf "/usr/include/qcustomplot.h"  "${srcdir}/${_pkgname}/src/qcustomplot.h"
}

build() {
    cd "${srcdir}/${_pkgname}/src/"
    qmake
    make
}

package() {
    install -dm0755 "${pkgdir}/usr/bin/" \
                   "${pkgdir}/usr/share/${pkgname}/"

    cd ${srcdir}/${_pkgname}/src/
#    cp --preserve=mode -r ${_pkgname} language config themes slave  "${pkgdir}/usr/share/${pkgname}/"

#    cd pkg/
#    cp --preserve=mode -r icons "${pkgdir}/usr/share/"

    #ln -sf  "/usr/share/${pkgname}/${_pkgname}" "${pkgdir}/usr/bin/${pkgname}"
#    install -Dm755 "${srcdir}/${pkgname}" "${pkgdir}/usr/bin/${pkgname}"
    install -Dm0644 "${srcdir}/${_pkgname}/LICENSE" "${pkgdir}/usr/share/licenses/${pkgname}/LICENSE"
#    install -Dm644 "${srcdir}/${_pkgname}-${pkgver}/${_pkgname}/pkg/io.github.skiars.${pkgname}.desktop" "${pkgdir}/usr/share/applications/io.github.skiars.${pkgname}.desktop"
#    install -Dm644 "${srcdir}/${_pkgname}-${pkgver}/${_pkgname}/pkg/20-usb-serial.rules" "${pkgdir}/etc/udev/rules.d/20-usb-serial.rules"
#    install -Dm644 "${srcdir}/${_pkgname}-${pkgver}/${_pkgname}/pkg/io.github.skiars.${pkgname}.metainfo.xml" "${pkgdir}/usr/share/metainfo/io.github.skiars.${pkgname}.metainfo.xml"
}

```

```bash
❯ makepkg -sf                                    
==> 正在创建软件包：serialtest-git 0.1.2-1 (2022年02月17日 星期四 21时00分59秒)
==> 正在检查运行时依赖关系...
[... 22 lines cut ...]
/usr/bin/uic ui/controlitem.ui -o ui_controlitem.h
/usr/bin/uic ui/mainwindow.ui -o ui_mainwindow.h
/usr/bin/uic ui/plottab.ui -o ui_plottab.h
g++ -c -pipe -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -std=gnu++11 -flto -fno-fat-lto-objects -Wall -Wextra -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -I/usr/include/qt -I/usr/include/qt/QtPrintSupport -I/usr/include/qt/QtWidgets -I/usr/include/qt/QtGui -I/usr/include/qt/QtSerialPort -I/usr/include/qt/QtCore -I. -I. -I/usr/lib/qt/mkspecs/linux-g++ -o main.o main.cpp
g++ -c -pipe -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -std=gnu++11 -flto -fno-fat-lto-objects -Wall -Wextra -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -I/usr/include/qt -I/usr/include/qt/QtPrintSupport -I/usr/include/qt/QtWidgets -I/usr/include/qt/QtGui -I/usr/include/qt/QtSerialPort -I/usr/include/qt/QtCore -I. -I. -I/usr/lib/qt/mkspecs/linux-g++ -o mycustomplot.o mycustomplot.cpp
g++ -c -pipe -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -Wp,-D_GLIBCXX_ASSERTIONS -std=gnu++11 -flto -fno-fat-lto-objects -Wall -Wextra -D_REENTRANT -fPIC -DQT_DEPRECATED_WARNINGS -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -I/usr/include/qt -I/usr/include/qt/QtPrintSupport -I/usr/include/qt/QtWidgets -I/usr/include/qt/QtGui -I/usr/include/qt/QtSerialPort -I/usr/include/qt/QtCore -I. -I. -I/usr/lib/qt/mkspecs/linux-g++ -o mysettings.o mysettings.cpp
make: *** 没有规则可制作目标“qcustomplot.cpp”，由“qcustomplot.o” 需求。 停止。
make: *** 正在等待未完成的任务....
In file included from mysettings.cpp:1:
mysettings.h: 在成员函数‘MySettings& MySettings::operator=(const MySettings&)’中:
mysettings.h:25:47: 警告：在有返回值的函数中未发现 return 语句 [-Wreturn-type]
   25 |     MySettings& operator=(const MySettings&) {}
      |                                               ^
      |                                               return *this;
In file included from mainwindow.h:23,
                 from main.cpp:1:
mysettings.h: 在成员函数‘MySettings& MySettings::operator=(const MySettings&)’中:
mysettings.h:25:47: 警告：在有返回值的函数中未发现 return 语句 [-Wreturn-type]
   25 |     MySettings& operator=(const MySettings&) {}
      |                                               ^
      |                                               return *this;
==> 错误： 在 build() 中发生一个错误。
    正在放弃...

```

wh201906, 2022-02-17:
arch linux有默认图标之类的吗？一时半会儿想不出什么好的图标来

taotieren, 2022-02-17:
[image]
桌面文件创建好了但是没有图标
还有一个问题是 串口设备的 rules 文件

wh201906, 2022-02-17:
rules文件是什么情况？

taotieren, 2022-02-17:
> rules文件是什么情况？

```bash
serialtest-git on  master [?] 
❯ cat /etc/udev/rules.d/99-xfel.rules 
# Copy this file to /etc/udev/rules.d/ or /usr/lib/udev/rules.d/
# If rules fail to reload automatically, you can refresh udev rules
# with the command "sudo udevadm control --reload"

ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="1f3a", ATTRS{idProduct}=="efe8", MODE="666", GROUP="users" TEST=="power/autosuspend", ATTR{power/autosuspend}="-1"

serialtest-git on  master [?] 
❯ cat /etc/udev/rules.d/10-h7tool.rules 
# Copy this file to /etc/udev/rules.d/ or /usr/lib/udev/rules.d/
# If rules fail to reload automatically, you can refresh udev rules
# with the command "sudo udevadm control --reload"

ACTION!="add|change", SUBSYSTEM!="usb_device", GOTO="h7tool_rules_end"

# SUBSYSTEM=="gpio", MODE="0666", GROUP="plugdev", TAG+="uaccess"

# SUBSYSTEM!="usb|tty|hidraw", GOTO="h7tool_rules_end"

# Please keep this list sorted by VID:PID

# H7-tool
# ATTRS{idVendor}=="c251", ATTRS{idProduct}=="f00a", MODE="666", GROUP="plugdev", TAG+="uaccess"
# BUS=='usb',
ATTRS{manufacturer}=="H7-TOOL By ARMFLY", MODE="0666", GROUP="plugdev", TAG+="uaccess"

LABEL="h7tool_rules_end"

serialtest-git on  master [?] 
❯ cat /etc/udev/rules.d/99-nuclei.rules 
# Copy this file to /etc/udev/rules.d/
# If rules fail to reload automatically, you can refresh udev rules
# with the command "udevadm control --reload"

# This rules are based on the udev rules from the OpenOCD project, with unsupported probes removed.
# See http://openocd.org/ for more details.
#
# This file is available under the GNU General Public License v2.0

ACTION!="add|change", GOTO="nuclei_rules_end"

#SUBSYSTEM=="gpio", MODE="0660", GROUP="plugdev", TAG+="uaccess"

SUBSYSTEM!="usb|tty|hidraw", GOTO="nuclei_rules_end"

# [GNU MCU Eclipse] -----------------------------------------------------------
# To simplify access, the access rights were changed from:
#   MODE="660", GROUP="plugdev", TAG+="uaccess"
# to:
#   MODE="666"
# -----------------------------------------------------------------------------

# Please keep this list sorted by VID:PID

# nuclei 
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6010", MODE="666", GROUP="plugdev"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="666", GROUP="plugdev"

LABEL="nuclei_rules_end"
serialtest-git on  master [?] 
❯ cat /etc/udev/rules.d/99-xfel.rules 
# Copy this file to /etc/udev/rules.d/ or /usr/lib/udev/rules.d/
# If rules fail to reload automatically, you can refresh udev rules
# with the command "sudo udevadm control --reload"

ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}=="1f3a", ATTRS{idProduct}=="efe8", MODE="666", GROUP="users" TEST=="power/autosuspend", ATTR{power/autosuspend}="-1"

serialtest-git on  master [?] 
❯ cat /etc/udev/rules.d/10-h7tool.rules 
# Copy this file to /etc/udev/rules.d/ or /usr/lib/udev/rules.d/
# If rules fail to reload automatically, you can refresh udev rules
# with the command "sudo udevadm control --reload"

ACTION!="add|change", SUBSYSTEM!="usb_device", GOTO="h7tool_rules_end"

# SUBSYSTEM=="gpio", MODE="0666", GROUP="plugdev", TAG+="uaccess"

# SUBSYSTEM!="usb|tty|hidraw", GOTO="h7tool_rules_end"

# Please keep this list sorted by VID:PID

# H7-tool
# ATTRS{idVendor}=="c251", ATTRS{idProduct}=="f00a", MODE="666", GROUP="plugdev", TAG+="uaccess"
# BUS=='usb',
ATTRS{manufacturer}=="H7-TOOL By ARMFLY", MODE="0666", GROUP="plugdev", TAG+="uaccess"

LABEL="h7tool_rules_end"

serialtest-git on  master [?] 
❯ cat /etc/udev/rules.d/99-nuclei.rules 
# Copy this file to /etc/udev/rules.d/
# If rules fail to reload automatically, you can refresh udev rules
# with the command "udevadm control --reload"

# This rules are based on the udev rules from the OpenOCD project, with unsupported probes removed.
# See http://openocd.org/ for more details.
#
# This file is available under the GNU General Public License v2.0

ACTION!="add|change", GOTO="nuclei_rules_end"

#SUBSYSTEM=="gpio", MODE="0660", GROUP="plugdev", TAG+="uaccess"

SUBSYSTEM!="usb|tty|hidraw", GOTO="nuclei_rules_end"

# [GNU MCU Eclipse] -----------------------------------------------------------
# To simplify access, the access rights were changed from:
#   MODE="660", GROUP="plugdev", TAG+="uaccess"
# to:
#   MODE="666"
# -----------------------------------------------------------------------------

# Please keep this list sorted by VID:PID

# nuclei 
SUBSYSTEM=="usb", ATTR{idVendor}=="0403", ATTR{idProduct}=="6010", MODE="666", GROUP="plugdev"
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6010", MODE="666", GROUP="plugdev"

LABEL="nuclei_rules_end"

```

taotieren, 2022-02-17:
反馈一个 bug
连续单击 `打开` 两次或两次以上
[image]
[image]

taotieren, 2022-02-17:
运行的时候会生成 preference.ini 配置文件
能指定生成位置吗？如 
/usr/share/serialtest/preference.ini

wh201906, 2022-02-17:
刚才你说的bug是指没有做中文翻译吗？我会在下一个版本当中加上去

wh201906, 2022-02-17:
目前preference.ini文件的生成位置就是在程序的运行目录下

taotieren, 2022-02-17:
> 目前 `preference.ini` 文件的生成位置就是在程序的运行目录下

我将 `SerialTest` 安装到 `/usr/share/serialtest/`
并创建一个 serialtest 文件，内容如下：
```bash
install -Dm0755 /dev/stdin /usr/bin/serialtest << EOF
#!/bin/bash
cd /usr/share/serialtest/
./SerialTest
EOF
```
运行后并不会在 `/usr/share/serialtest/` 目录下生成 `preference.ini` 文件 也没有错误提示

wh201906, 2022-02-17:
因为运行目录是在~/下，所以preferences.ini也是在~/下产生的

taotieren, 2022-02-17:
开发版的测试 PKGBUILD
```bash
# Maintainer: taotieren <admin@taotieren.com>

_pkgname=SerialTest
pkgname=serialtest-git
pkgver=0.1.2.r5.g72df387
pkgrel=1
pkgdesc="A cross-platform serial port test tool."
arch=('any')
url="https://github.com/wh201906/SerialTest"
license=('GPL2.1')
provides=(${_pkgname})
conflicts=(${pkgname%-git})
#replaces=(${pkgname})
depends=('qcustomplot' 'qt5-serialport')
makedepends=('qt5-tools' 'git')
backup=()
options=('!strip')
install=${pkgname}.install
source=("${_pkgname}::git+${url}.git"
        "${pkgname}.install")
sha256sums=('SKIP'
            '303f34246c0d341e1093d6e486e2cbfdbbb0d122d751de649f562ebac37777b4')

prepare() {
    cd "${srcdir}/${_pkgname}/"
    git checkout dev
}

pkgver() {
    cd "${srcdir}/${_pkgname}/"
    git describe --long --tags | sed 's/V//g;s/\([^-]*-g\)/r\1/;s/-/./g'
}

build() {
    cd "${srcdir}/${_pkgname}/src/"
    qmake
    make
}

package() {
    install -Dm0755 "${srcdir}/${_pkgname}/src/${_pkgname}"  "${pkgdir}/usr/share/${pkgname%-git}/${_pkgname}"

    install -Dm0755 /dev/stdin "${pkgdir}/usr/bin/${pkgname%-git}" << EOF
#!/bin/bash
cd /usr/share/serialtest/
./SerialTest "\$@"
EOF
      install -Dm0777 /dev/stdin "${pkgdir}/usr/share/${pkgname%-git}/preference.ini" << EOF

EOF

    install -Dm0644 /dev/stdin "${pkgdir}/usr/share/metainfo/io.github.wh201906.serialtest.metainfo.xml" << EOF
<?xml version="1.0" encoding="UTF-8"?>
<component type="desktop-application">
  <id>io.github.wh201906.serialtest</id>

  <name>SerialTest</name>
  <summary>SerialTest</summary>

  <metadata_license>MIT</metadata_license>
  <project_license>GPL-2.0-or-later</project_license>

  <description>
    <p>
      A cross-platform serial port test tool.
    </p>
  </description>

  <launchable type="desktop-id">io.github.wh201906.serialtest.desktop</launchable>
</component>
EOF

    install -Dm0644 /dev/stdin "${pkgdir}/usr/share/applications/io.github.wh201906.serialtest.desktop" << EOF
[Desktop Entry]
Version=1.0
Type=Application

Name=SerialTest
Comment=SerialTest
Categories=Development;Electronics;

Icon=serialtest
Exec=serialtest
Terminal=false
EOF

    install -Dm0644 "${srcdir}/${_pkgname}/LICENSE" "${pkgdir}/usr/share/licenses/${pkgname}/LICENSE"
#    install -Dm644 "${srcdir}/${_pkgname}-${pkgver}/${_pkgname}/pkg/20-usb-serial.rules" "${pkgdir}/etc/udev/rules.d/20-usb-serial.rules"
}

```

taotieren, 2022-02-17:
创建空文件放在哪里也不行，`/usr/share/serialtest` 没有权限进行 `写` 操作
Linux 下的建议做法是用户的配置文件用户的目录 `~/.config/SerialTest/` 下，
```bash
/usr/share/serialtest🔒 
❯ ls -lash                                 
总用量 388K
4.0K drwxr-xr-x   2 root root 4.0K  2月 17 22:14 .
 12K drwxr-xr-x 367 root root  12K  2月 17 22:08 ..
4.0K -rwxrwxrwx   1 root root    1  2月 17 22:14 preference.ini
368K -rwxr-xr-x   1 root root 367K  2月 17 22:14 SerialTest
```

wh201906, 2022-02-17:
那有点麻烦了
我尽量下个版本加个设置界面，允许用户改变配置文件的位置。这样应该能解决问题。

taotieren, 2022-02-17:
> 那有点麻烦了 我尽量下个版本加个设置界面，允许用户改变配置文件的位置。这样应该能解决问题。

直接默认配置到 `~/.config/SerialTest/` 目录是最好，最通用的做法

wh201906, 2022-02-17:
因为还有Windows端。我之前设计的时候是想着配置文件和可执行文件放在一起，然后可以免安装使用，也不会对注册表或者其它目录造成影响

taotieren, 2022-02-17:
开发版已经提交到 ArchLinux 的 AUR 仓库 https://aur.archlinux.org/packages/serialtest-git
发行版还有问题没处理，暂时编译过不了就没提交了

wh201906, 2022-02-17:
好的，十分感谢！

taotieren, 2022-02-17:
> 因为还有Windows端。我之前设计的时候是想着配置文件和可执行文件放在一起，然后可以免安装使用，也不会对注册表或者其它目录造成影响

可以参考 Arch 的打包规则

[https://wiki.archlinux.org/title/Arch_package_guidelines](https://wiki.archlinux.org/title/Arch_package_guidelines)

wh201906, 2022-02-17:
好的

taotieren, 2022-02-17:
> 好的，十分感谢！

发行版的 PKGBUILD 也写好了，等找到 图标 和修复配置文件的问题后在上传到 AUR 那边

wh201906, 2022-02-17:
好的，等我做好之后告诉你

wh201906, 2022-08-20:
@taotieren 现在图标和配置文件都解决了，src/icon/icon.png是原始图片。
目前在Ubuntu上编译还需要安装一个qtconnectivity5-dev来支持蓝牙。
如果方便的话还望能够再帮忙打包一下，谢谢！

taotieren, 2022-08-20:
OK，晚点测试下

wh201906, 2022-08-20:
如果安装脚本能够在安装完成之后在~/.config/SerialTest/目录下新建一个preference.ini，那么程序就会使用此文件作为配置文件

taotieren, 2022-08-20:
> 如果安装脚本能够在安装完成之后在~/.config/SerialTest/目录下新建一个preference.ini，那么程序就会使用此文件作为配置文件

```bash
post_install() {
  udevadm trigger
  udevadm control --reload-rules
}

check_preference() {
  if [ ! -f $HOME/.config/SerialTest/preference.ini ]; then
    touch $HOME/.config/SerialTest/preference.ini
  fi
}

remove_preference() {
  if [ -f $HOME/.config/SerialTest/preference.ini ]; then
    rm -rf $HOME/.config/SerialTest
  fi
}

post_upgrade() {
  post_install
  remove_preference
  check_preference
}

post_remove() {
  post_install
  remove_preference
}

```

taotieren, 2022-08-20:
> @taotieren 现在图标和配置文件都解决了，src/icon/icon.png是原始图片。 目前在Ubuntu上编译还需要安装一个qtconnectivity5-dev来支持蓝牙。 如果方便的话还望能够再帮忙打包一下，谢谢！
 arch 的包是 qt5-connectivity
```bash
❯ yay -Qi qt5-connectivity      
名字           : qt5-connectivity
版本           : 5.15.5+kde+r5-1
描述           : Provides access to Bluetooth hardware
架构           : x86_64
URL            : https://www.qt.io
软件许可       : GPL3
                 LGPL3
                 FDL
                 custom
组             : qt
                 qt5
提供           : 无
依赖于         : qt5-base
                 bluez-libs
可选依赖       : qt5-declarative: QML bindings [已安装]
依赖它         : serial-studio
被可选依赖     : python-pyqt5
与它冲突       : 无
取代           : 无
安装后大小     : 2033.17 KiB
打包者         : Antonio
                 Rojas
                 <arojas@archlinux.org>
编译日期       : 2022年06月18日 星期六 21时12分46秒
安装日期       : 2022年06月19日 星期日 18时13分57秒
安装原因       : 作为其他软件包的依赖关系安装
安装脚本       : 否
验证者         : 数字签名

yay -Ql qt5-connectivity
qt5-connectivity /usr/
qt5-connectivity /usr/bin/
qt5-connectivity /usr/bin/sdpscanner
qt5-connectivity /usr/include/
qt5-connectivity /usr/include/qt/
qt5-connectivity /usr/include/qt/QtBluetooth/
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/adapter1_bluez5_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/adapter_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/agent_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/androidbroadcastreceiver_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/battery1_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/bluetoothmanagement_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/bluez5_helper_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/bluez_data_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/btdelegates_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/btraii_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/device1_bluez5_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/device_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/devicediscoverybroadcastreceiver_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/dummy_helper_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/gattchar1_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/gattdesc1_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/gattservice1_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/hcimanager_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/inputstreamthread_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/jni_android_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/lecmaccalculator_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/localdevicebroadcastreceiver_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/lowenergynotificationhub_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/manager_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/obex_agent_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/obex_client1_bluez5_p.h
qt5-connectivity /usr/include/qt/QtBluetooth/5.15.5/QtBluetooth/private/obex_client_p.h
[... 210 lines cut ...]
qt5-connectivity /usr/lib/libQt5Nfc.so.5.15.5
qt5-connectivity /usr/lib/pkgconfig/
qt5-connectivity /usr/lib/pkgconfig/Qt5Bluetooth.pc
qt5-connectivity /usr/lib/pkgconfig/Qt5Nfc.pc
qt5-connectivity /usr/lib/qt/
qt5-connectivity /usr/lib/qt/mkspecs/
qt5-connectivity /usr/lib/qt/mkspecs/modules/
qt5-connectivity /usr/lib/qt/mkspecs/modules/qt_lib_bluetooth.pri
qt5-connectivity /usr/lib/qt/mkspecs/modules/qt_lib_bluetooth_private.pri
qt5-connectivity /usr/lib/qt/mkspecs/modules/qt_lib_nfc.pri
qt5-connectivity /usr/lib/qt/mkspecs/modules/qt_lib_nfc_private.pri
qt5-connectivity /usr/lib/qt/qml/
qt5-connectivity /usr/lib/qt/qml/QtBluetooth/
qt5-connectivity /usr/lib/qt/qml/QtBluetooth/libdeclarative_bluetooth.so
qt5-connectivity /usr/lib/qt/qml/QtBluetooth/plugins.qmltypes
qt5-connectivity /usr/lib/qt/qml/QtBluetooth/qmldir
qt5-connectivity /usr/lib/qt/qml/QtNfc/
qt5-connectivity /usr/lib/qt/qml/QtNfc/libdeclarative_nfc.so
qt5-connectivity /usr/lib/qt/qml/QtNfc/plugins.qmltypes
qt5-connectivity /usr/lib/qt/qml/QtNfc/qmldir
qt5-connectivity /usr/share/
qt5-connectivity /usr/share/licenses/
qt5-connectivity /usr/share/licenses/qt5-connectivity

```

taotieren, 2022-08-20:
干净的编译环境测试
```bash
serialtest on  master [!?] took 24s 
❯ extra-x86_64-build
:: Synchronizing package databases...
 core downloading...
 extra downloading...
 community downloading...
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...

warning: insufficient columns available for table display
Packages (39) archlinux-keyring-20220727-1
              binutils-2.39-2
              ca-certificates-mozilla-3.82-1
              curl-7.84.0-2
              e2fsprogs-1.46.5-4
              elfutils-0.187-2
              gcc-12.1.1-4
              gcc-libs-12.1.1-4
              glib2-2.72.3-1
              glibc-2.36-2
              gnupg-2.2.36-1
              gnutls-3.7.7-3
              gpgme-1.18.0-1
              jansson-2.14-2
              libarchive-3.6.1-2
              libcap-2.65-1
              libelf-0.187-2
              libevent-2.1.12-2
              libidn2-2.3.3-1
              libisl-0.25-1
              libldap-2.6.3-1
              libnghttp2-1.48.0-1
              libsysprof-capture-3.44.0-2
              libtirpc-1.3.3-1
              libtool-2.4.7-3
              linux-api-headers-5.18.15-1
              mpfr-4.1.0.p13-3
              nettle-3.8.1-1
              openssl-1.1.1.q-1
              pacman-6.0.1-7
              pacman-mirrorlist-20220724-1
              sqlite-3.39.2-2
              sudo-1.9.11.p3-1
              systemd-libs-251.4-1
              tzdata-2022c-1
              util-linux-2.38.1-1
              util-linux-libs-2.38.1-1
              xz-5.2.6-1
              zstd-1.5.2-7

Total Download Size:   122.32 MiB
Total Installed Size:  502.54 MiB
Net Upgrade Size:       27.53 MiB

:: Proceed with installation? [Y/n] 
:: Retrieving packages...
 gcc-12.1.1-4-x86_64 downloading...
 gcc-libs-12.1.1-4-x86_64 downloading...
 glibc-2.36-2-x86_64 downloading...
 binutils-2.39-2-x86_64 downloading...
 openssl-1.1.1.q-1-x86_64 downloading...
 glib2-2.72.3-1-x86_64 downloading...
 util-linux-2.38.1-1-x86_64 downloading...
 gnupg-2.2.36-1-x86_64 downloading...
 sudo-1.9.11.p3-1-x86_64 downloading...
 sqlite-3.39.2-2-x86_64 downloading...
 e2fsprogs-1.46.5-4-x86_64 downloading...
[... 819 lines cut ...]
installing libnsl...
installing python...
Optional dependencies for python
    python-setuptools
    python-pip
    sqlite [installed]
    mpdecimal: for decimal
    xz: for lzma [installed]
    tk: for tkinter
installing pyalpm...
installing licenses...
installing python-pyelftools...
installing namcap...
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
Checking PKGBUILD
Checking serialtest-0.2-0-any.pkg.tar.zst
serialtest E: ELF file ('usr/bin/serialtest') found in an 'any' package.
serialtest W: ELF file ('usr/bin/serialtest') is unstripped.
serialtest E: GPL2.1 is not a common license (it's not in /usr/share/licenses/common/)
==> Running checkpkg
error: target not found: serialtest
==> WARNING: Skipped checkpkg due to missing repo packages

```

taotieren, 2022-08-20:
[image]
运行时的配置路径还是有问题，
在编译脚本中对 `$HOME` 操作不行，需要对 `/home/$USER` 下处理。需要你程序上处理。
```bash
serialtest on  master [!?] took 59s 
❯ yay -U serialtest-0.2-0-any.pkg.tar.zst 
正在加载软件包...
正在解析依赖关系...
正在查找软件包冲突...

软件包 (1) serialtest-0.2-0

全部安装大小：  0.84 MiB
净更新大小：    0.44 MiB

:: 进行安装吗？ [Y/n] 
(1/1) 正在检查密钥环里的密钥                       [##########################] 100%
(1/1) 正在检查软件包完整性                         [##########################] 100%
(1/1) 正在加载软件包文件                           [##########################] 100%
(1/1) 正在检查文件冲突                             [##########################] 100%
:: 正在处理软件包的变化...
(1/1) 正在更新 serialtest                          [##########################] 100%
touch: 无法 touch '/root/.config/SerialTest/preference.ini': 没有那个文件或目录
错误：命令未能被正确执行
:: 正在运行事务后钩子函数...
(1/3) Arming ConditionNeedsUpdate...
(2/3) Refreshing PackageKit...
(3/3) Updating the desktop file MIME type cache...

```

wh201906, 2022-08-20:
你点一下“在系统默认配置文件夹中创建配置文件”，然后删掉/home/<username>/preference.ini试试？

taotieren, 2022-08-21:
> 你点一下“在系统默认配置文件夹中创建配置文件”，然后删掉/home//preference.ini试试？

[image]
选择默认位置创建后会生存相应的文件，但还是会从`运行处` 新建配置文件，在什么位置运行 `serialtest` 就会在相应的位置创建配置文件，哪怕是从 `~/.config/serialtest/` 下导入配置文件，重启后还是`运行处`位置的配置文件，不能指定配置文件的路径。

wh201906, 2022-08-21:
我在windows环境下运行没有这个问题。程序启动时如果当前目录下没有preference.ini，而系统默认配置文件夹(Linux下为~/.config/SerialTest)中有preference.ini，那么程序就会加载后者，不会新建前者。
我不知道是不是Linux系统或者Arch Linux的问题，等我先起个Ubuntu环境试试

wh201906, 2022-08-21:
Ubuntu下测试是正常的

taotieren, 2022-08-21:
https://aur.archlinux.org/packages/serialtest

https://aur.archlinux.org/packages/serialtest-git

wh201906, 2022-08-21:
现在能在Arch Linux上正常运行了吗？

taotieren, 2022-08-22:
> 现在能在Arch Linux上正常运行了吗？

这个问题还是存在，有点奇怪，你那边打包的软件包能发一下吗？我解包运行试试看。

wh201906, 2022-08-22:
我是直接编译运行的

wh201906, 2022-08-22:
@taotieren 我自己弄了个Manjaro的环境，然后用yay安装之后打开程序，在系统默认配置路径生成配置文件，关闭应用后删除当前目录下的配置文件(直接用rm删掉)，然后重启应用，程序会使用新的配置文件  

[image]

taotieren, 2022-08-22:
可能是我没有删除本地的配置文件导致。

wh201906, 2022-08-22:
那问题就算是都解决了？
如果感兴趣的话，你可以fork一份master分支，然后把arch linux上安装的命令写进去再提交一个PR

taotieren, 2022-08-23:
还有几个文件需要 PR 上来，桌面启动文件和 xml 描述文件，这种放源码中什么位置合适呢？没有的话我会新建一个目录存放。

wh201906, 2022-08-23:
要不具体细节用邮件商讨？

commit daa78b1e (2022-08-20): Translations and logo

Fixes #2
closed 2022-08-23 by wh201906