wh201906/SerialTest #3: dynamic memory Arduino UNO R3

bsbhro, 2022-03-04:
Hi, I built the program flawlessly. I receive data (blink.ino with serial "on" and "off" :). Now I would like to plot your plot_cls.ino . But this compilation error comes up:
Global variables use 2251 bytes (109%) of dynamic memory, leaving -203 bytes for local variables. Maximum is 2048 bytes.
Not enough memory; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing your footprint.
Compilation error: data section exceeds available space in board}

I'm not good enough to change your sketch. Did you use another Arduino?
Thank you in advance, Bernard

wh201906, 2022-03-04:
Hi.
The plot_cls.ino and plot_performance.ino generate the waveform table in the setup(), then print them in loop(). The table takes a lot of RAM. 
You can try plot_realtime.ino, which calculate the waveform in the loop. It is slightly slower, but takes much less RAM.
I use ESP32 so the RAM is pretty large(~300k available) and I didn't consider the RAM issue. I will change the demos soon.

wh201906, 2022-03-04:
You can use the new demo now.

bsbhro, 2022-03-04:
I tested the tree Arduino demos on the UNO R3 and an ESP8266 12-F. All fine.
When I start the ./SerialTest the first line at the Terminal is 'QCPGraph* QCustomPlot::graph(int) const index out of bounds: 0'. Is it relevant?
Very good and interesting work! Thank you very much!

wh201906, 2022-03-04:
Happy to help.
The "index out of bounds" bug do exists, but it doesn't matters. I might fix it when I want.

wh201906, 2022-03-04:
@bsbhro Would you like to try the [control_pwm](https://github.com/wh201906/SerialTest/tree/master/demo/Arduino/control_pwm) demo? 
It's convenient to send "on" and "off" in the Control(Shortcut) panel. You can even change the brightness of a LED if you connect it to a pin supporting PWM output.

bsbhro, 2022-03-04:
Interesting! On Arduino I measured values between 0.001 V (a000) and 4.952 V (a255), at PWM-Pin 5. On the NodeMCU at Pin D5 I got 0.001 V to 3.333 V.
As a newbie I hav

wh201906, 2022-03-05:
The typical VCC of Arduino UNO board is 5V and the typical VCC of NodeMCU board is 3.3V. 
When you set the analog value to 255, the pin is always at high level, which means that the output voltage is always VCC.
When you set the analog value to 0, the pin is always at low level, which means that the output voltage is always 0V.
When you set the analog value to any value between 0 and 255, the actual output is a square wave with different duty cycle. higher value means higher duty cycle.

wh201906, 2022-03-12:
Closing it for now. You can re-open it if necessary, or you can create a new issue if you meet other problems.

closed 2022-03-12 by wh201906