wh201906/SerialTest #7: Android 版本可否支持普通串口？

ZeaKyX, 2022-07-08:
主要是很多开发板的场景，能否指定 /dev/ttyXXX 端口？
这个工具跟桌面端的工具类似，可以查看已收和已发的数据量，非常方便测试。
指定串口的话，类似这种：https://github.com/GeekBugs/Android-SerialPort

wh201906, 2022-07-08:
因为程序使用的QSerialPort不支持安卓端，所以无法直接使用
目前dev分支支持TCP/UDP连接，你可以在开发版上安装一个UART转TCP的app作为中转，然后用网络来连接，也可以达到类似的效果
UART转TCP的app可以试一下[这个](https://play.google.com/store/apps/details?id=com.hardcodedjoy.tcpuart)

wh201906, 2023-05-14:
V0.3中支持形如`/dev/ttyXXX`的硬件串口，但需要在“设置”面板当中启用
我过几天会发布此版本

closed 2023-05-14 by wh201906

yangmeishu, 2024-05-21:
> ~因为程序使用的QSerialPort不支持安卓端，所以无法直接使用~
> 
> 对于/dev/ttyXXX串口：
> 
> 连接效果图
> ![图片](https://private-user-images.githubusercontent.com/62299611/237562526-34e028e4-65b5-4285-84fb-1e8c5ccf09b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYzMDQyNDIsIm5iZiI6MTcxNjMwMzk0MiwicGF0aCI6Ii82MjI5OTYxMS8yMzc1NjI1MjYtMzRlMDI4ZTQtNjViNS00Mjg1LTg0ZmItMWU4YzVjY2YwOWI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIxVDE1MDU0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTIyNzcxMGY1NTYxNDUzMDk5YjY1MzFhZDRhODhlNTQwNTdmN2ZhYjk0MWYzMzBmZmE1YmM2ZDYwOTBkMDVmOGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.74u-jW4JM04LloXEL0dlemwiaf0swJLvQHF9oQ5WUNE)
> 
> 可以指定/dev/ttyXXX串口，但是需要先用`chmod 777 /dev/ttyXXX`使APP有权限访问串口。而且目前的代码(V0.2.4)当中禁用了串口菜单，需要删除[这部分代码](https://github.com/wh201906/SerialTest/blob/V0.2.4/src/devicetab.cpp#L305-L307)手动编译才行。 我应该会在下一个版本中开启这个功能。
> 
> 对于USB转串口设备： QSerialPort不支持此类设备，所以无法直接使用。但你可以安装一个UART转TCP的APP作为中转，然后用TCP来连接，也可以达到类似的效果 UART转TCP的app可以试一下[这个](https://play.google.com/store/apps/details?id=com.hardcodedjoy.tcpuart)

作者你好，我是安卓usb转串口，ch340。打开后，没有串口。但是serial usb termial 能有。找到搜寻信息来到这里。你推荐的这个程序。和你写的另一个程序 [SerialPipe](https://github.com/wh201906/SerialPipe) 是实现同一个功能吗，有什么区别呢

wh201906, 2024-05-22:
> 对于USB转串口设备：
QSerialPort不支持此类设备，所以无法直接使用。但你可以安装一个UART转TCP的APP作为中转，然后用TCP来连接，也可以达到类似的效果
UART转TCP的app可以试一下[这个](https://play.google.com/store/apps/details?id=com.hardcodedjoy.tcpuart)

目前SerialTest暂不支持直连usb转串口设备，需要用一个中转软件把USB串口先转成TCP或者UDP，然后才能用SerialTest连接。SerialPipe和上面提到的TCPUART都是用来做中转的