wh201906/SerialTest #20: QBluetoothLocalDevice::allDevices() 获取设备为0，无法获取到本地蓝牙设备信息

yanliangyy, 2023-01-10:
您好 可爱的开发者
      我使用qt 5.15.2 用源码打包出来，无法获取到本地蓝牙设备，而我我本地是有蓝牙设备的。跟踪代码发现是QBluetoothLocalDevice::allDevices() 获取设备为0。请问是不是还需要其他第三方的库，或者其他支持。
程序截图如下：
[image]

wh201906, 2023-01-10:
不需要其它第三方库。你的蓝牙适配器打开了吗？如果没打开或者没有适配器蓝牙相关的功能就会显示不可用

wh201906, 2023-01-10:
如果你有qt的开发环境，你可以试着运行一下其它qt蓝牙的例程，看看能否在你的机器上正常运行

yanliangyy, 2023-01-11:
> 
我看你打出的V0.2.2-win32.7z包里面多了几个dll，我用你打好的release包：V0.2.2-win32 没有这样的问题，可以正常显示，只是我打出来的包不能获取到，这就很奇怪，我们的环境有什么区别？
如下：
[image]

yanliangyy, 2023-01-11:
> 如果你有qt的开发环境，你可以试着运行一下其它qt蓝牙的例程，看看能否在你的机器上正常运行
他自带的示例也不行

[image]

yanliangyy, 2023-01-11:
> 不需要其它第三方库。你的蓝牙适配器打开了吗？如果没打开或者没有蓝牙适配器，那么蓝牙相关的功能就会显示不可用

[image]

wh201906, 2023-01-11:
> 我看你打出的V0.2.2-win32.7z包里面多了几个dll，我用你打好的release包：V0.2.2-win32 没有这样的问题，可以正常显示，只是我打出来的包不能获取到，这就很奇怪，我们的环境有什么区别？

这几个dll是用mingw编译产生的exe所需的库文件，用windeployqt打包的时候不会打包这些文件，需要手动复制。
你的意思是V0.2.2-win32可以正常显示蓝牙设备吗？

yanliangyy, 2023-01-11:
> 

> > 我看你打出的V0.2.2-win32.7z包里面多了几个dll，我用你打好的release包：V0.2.2-win32 没有这样的问题，可以正常显示，只是我打出来的包不能获取到，这就很奇怪，我们的环境有什么区别？
> 
> 这几个dll是用mingw编译产生的exe所需的库文件，用windeployqt打包的时候不会打包这些文件，需要手动复制。 你的意思是V0.2.2-win32可以正常显示蓝牙设备吗？

是的 我直接下载的V0.2.2-win32 可以直接扫描出来  但是我自己编译的不行

wh201906, 2023-01-11:
你的qt环境是用在线安装包安装的吗？还是用源码手动编译安装的？

yanliangyy, 2023-01-11:
> 

> 你的qt环境是用在线安装包安装的吗？还是用源码手动编译安装的？

我使用的在线安装包，安装的5.15.2，以下是我的配置信息。请问你使用的是什么版本，你的配置信息方便截图出来吗？
[image]

yanliangyy, 2023-01-11:
感谢你的支持，我已经解决问题了，我用到是MSVC编译，所以..

wh201906, 2023-01-11:
按理说msvc编译也应该不至于蓝牙用不了。不过因为我没有msvc的环境，没法测试

yanliangyy, 2023-01-13:
> 按理说msvc编译也应该不至于蓝牙用不了。不过因为我没有msvc的环境，没法测试

实际的测试确实是msvc的支持要要差些，网上还说不能用MinGW编译蓝牙功能，这才使得我使用了msvc，结果发现扫描不到本地蓝牙

wh201906, 2023-01-13:
所以你用mingw编译之后能用了吗？

wh201906, 2023-04-15:
如果还有问题，你可以重新打开这个issue或者新开一个

closed 2023-04-15 by wh201906

wh201906, 2024-04-05:
我在 9760cccda87cd8bed31b83e744d615c728153771 当中添加了蓝牙功能对于MSVC编译器的支持，你可以试一下。

目前测下来MSVC编译出来的版本对BLE的支持最好，但是传统蓝牙还是MinGW编译出来的版本要好用些。

wh201906, 2024-04-18:
现在master分支也支持用MSVC编译蓝牙部分的代码了，对应的版本是V0.3.5