wh201906/SerialTest #23: 有关于绘图功能的详细文档吗？

Zheng-Bicheng, 2023-02-14:
作者你好，我想请问下有关于绘图功能的详细文档吗？绘图是根据哪个变量来绘制的？我这边把需要绘制的QByteArray传给rawReceivedData，其中数据为"100,100,100,100,100,100\n10,10,10,10,10,10\n......"。然后设置了帧分隔符“\n”，数据项分隔符","，发现绘图控件没有刷新。

wh201906, 2023-02-14:
目前有一个关于绘图功能如何使用的文档，主要是面向普通用户的
https://github.com/wh201906/SerialTest/blob/master/doc/tutorials/plot/plot_zh_CN.md
关于内部具体实现目前没有相关文档
以当前master分支上的最新提交https://github.com/wh201906/SerialTest/commit/717a69a7f628b3bdcd4ef2eaf25569e008fef0f1 为例：
`rawReceivedData`只是用来存储已接收的数据，用于“数据”选项卡当中的16进制显示和导出功能
程序中UI的更新是通过`MainWindow::updateRxUI()`来完成的，当中调用了`plotTab->newData(RxUIBuf)`来给“绘图”选项卡提供数据。

Zheng-Bicheng, 2023-02-14:
我修改了RxUIBuf的数值，发现绘图功能已经能成功绘制图像了。这里有还一个问题，我需要绘制的图像是一个连续的sin波形，也就是说传入RxUIBuf的数据将会是256个连续的数据，我要怎么修改让绘图能够成功识别呢？按照数据分隔符","来分割的话，256个连续数据会被当成256个不同的图例数据吧？

Zheng-Bicheng, 2023-02-14:
搞定了，发送数据的时候把分隔符全改成数据帧分隔符就行。感谢大佬的帮助。

closed 2023-02-14 by Zheng-Bicheng