wh201906/SerialTest #24: 有Android的编译教程吗？

Zheng-Bicheng, 2023-02-15:
有编译安卓程序的QT配置文档吗？这边PC已经跑通了，但是编译安卓就是通不过去，已经换了3个不同的QT版本了，能否指导一下。

wh201906, 2023-02-15:
有什么报错信息吗？我用Qt官方的5.15.2在线安装包安装的Android套件，直接编译不会报错
如果是在Linux下编译，你可以参考app在F-Droid上的自动编译步骤
https://gitlab.com/fdroid/fdroiddata/-/blob/master/metadata/priv.wh201906.serialtest.yml

Zheng-Bicheng, 2023-02-15:
> 

你是Windows系统吗？有办法打包一下你使用的SDK吗？使用的JDK版本号是多少。

wh201906, 2023-02-15:
SDK估计不太好打包

<details>
<summary>当前配置</summary>

[image]
[image]

</details>

你能够编译Qt自带的例程吗？这个app的Android配置不复杂，如果其它程序能编译这个app理论上也可以

Zheng-Bicheng, 2023-02-16:
> 

我测试一下哈，万分感谢大佬

closed 2023-03-08 by Zheng-Bicheng