wh201906/SerialTest #25: 安卓编译

Q356646, 2023-04-10:
你好大佬。编译安卓，需要哪个版本的qt版本来支持

wh201906, 2023-04-10:
5.15的任意版本应该都可以

Q356646, 2023-04-10:
[image]
你好！我的qt版本是5.15.2使用最新的0.2.2版本编译。出现图示的问题。这个是绘图那个需要安装什么包吗？由于没有怎么使用过曲线绘图，这个编译错误应该怎么处理

Q356646, 2023-04-10:
[image]

Q356646, 2023-04-10:
你好！我的qt版本是5.15.2使用最新的0.2.2版本编译。出现图示的问题。这个是绘图那个需要安装什么包吗？由于没有怎么使用过曲线绘图，这个编译错误应该怎么处理
[image]
[image]

wh201906, 2023-04-10:
你可以参考一下这里的编译步骤
https://github.com/wh201906/SerialTest/blob/master/doc/README/README_zh_CN.md#3-%E9%80%89%E6%8B%A9%E5%A6%82%E4%BD%95%E4%BD%BF%E7%94%A8qcustomplot
你需要去QCustomPlot的官网上下载.cpp和.h文件，然后放到src文件夹下面，再编译

zj2023410, 2023-04-10:
好的！谢谢解答

closed 2023-04-15 by wh201906