wh201906/SerialTest #27: Translation license

comradekingu, 2023-05-14:
https://hosted.weblate.org/projects/serialtest/app/#information says "proprietary",
but it should say `LGPLv3-or-later` like https://github.com/wh201906/SerialTest/blob/master/LICENSE

It can be changed in
https://hosted.weblate.org/settings/serialtest/app/
and
https://hosted.weblate.org/settings/serialtest/glossary/

:)

wh201906, 2023-05-14:
Thank you for point this out!
I'm not ready to use Weblate with this project now. Plus, I'm preparing changing the license of this project.
https://github.com/wh201906/SerialTest/discussions/26
When everything is ready, I will update the Weblate page to keep things fresh.

wh201906, 2023-05-27:
Fixed. Thanks!

closed 2023-05-27 by wh201906

wh201906, 2023-06-02:
@comradekingu Sorry. I accidentally deleted the translation of `nb_NO` when configuring Weblate. Would you like to start it again?

comradekingu, 2023-06-02:
@wh201906 
Will do when I get around to it. All the components are added for "nb_NO".