wh201906/SerialTest #29: Dark Theme

xmha97, 2023-05-17:
Please add the dark theme to this app.

wh201906, 2023-05-17:
It does have dark theme.
1. Go to "settings"
[image]
2. Select "Dark" in Appearance->Theme
[image]
3. Click "Set" behind the Theme selector
[image]
4. Restart the app
[image]

xmha97, 2023-05-18:
Thanks, now I need another feature:
- Please follow the system theme.

Like other apps:
[image]
Please add the "System" item.
[image]

wh201906, 2023-05-19:
It might be a little bit hard to implement it on Qt5. I'll try it when I'm free.

xmha97, 2023-06-11:
Please change "(None)" to "Follow system"
It works on Ubuntu fine.

[image]

[image]

wh201906, 2023-06-11:
@xmha97 It's a cross-platform app and the behavior is not the same on different platforms. So I think it's not good to change the `(None)` to `Follow System`.

wh201906, 2023-06-11:
Plus, I think I might add more themes in the future, for both dark style and light style. However, users cannot choose which dark style to use with a single `Follow System` option, when the system is in dark mode. 
Users can change the theme without restarting the app now (see 4d15a2fa4833201a8c33c995f05113ed64f5c795 and 7b3dccb753880fbbbeae32d0113625aad90c46f5). So I guess the feature is not so necessary now?

xmha97, 2023-06-12:
Thanks for the new version.

xmha97, 2023-06-12:
Android app does not follow OS theme, please reopen this issue.

wh201906, 2023-06-12:
Sorry but I don't want to add this feature for these reasons:
1. As I mentioned in https://github.com/wh201906/SerialTest/issues/29#issuecomment-1586198310, "Follow OS theme" is ambiguous when there are more than one dark theme. 
2. I guess you want this feature because its inconvenient for users to change the theme before V0.3.2. That's true, as they have to restart the app to do so. However it's less necessary now because I have removed the limitation. Following OS theme is good for the user experience, but it might cost a lot of time for me to implement this because it's a Qt app.
3. As you said in https://github.com/wh201906/SerialTest/issues/29#issuecomment-1586140024, the default theme does follow the OS theme on Ubuntu.

closed 2023-06-12 by wh201906