wh201906/SerialTest #32: Add monochrome icon

xmha97, 2023-06-12:
Please support the Android 13 "Themed App Icons" feature.

wh201906, 2023-06-12:
I found that some of the apps you listed doesn't have a specific configuration for the monochrome icon, and I don't know why the OS cannot change the icon into a monochrome one while others can. It will be much helpful if you could provide some information about it.

sebastien46, 2023-11-26:
> It will be much helpful if you could provide some information about it.

Here is the [official Android developer documentation](https://developer.android.com/develop/ui/views/launch/icon_design_adaptive) on the subject

wh201906, 2023-11-26:
@sebastien46 Thanks. It's useful.

sebastien46, 2023-12-05:
@wh201906 here's a mockup I made using inkscape:
[image]

I also saw that you didn't use the vector file for drawing the regular / adaptive app icon, so here's a version of that:

[image]

I would've tried making a PR but the QT Creator VCS tree is really too confusing for me 😄

wh201906, 2023-12-05:
@sebastien46 Thank you so much! I'll try the information you provided.

wh201906, 2024-01-27:
I finally got an Android 14 device to test it. This is implemented in commit 1e62c938d718d4f0be21d3e04ad45cff3d962afa and it's working on my device.

wh201906, 2024-02-26:
> When it'll implemented in new update

Ah sorry I didn't get the notification of this reply.
I'll release it in around 1~2 weeks.

saltsoftdrink, 2024-02-26:
> > When it'll implemented in new update
> 
> Ah sorry I didn't get the notification of this reply. I'll release it in around 1~2 weeks.

Thanks Man!

wh201906, 2024-03-10:
@saltsoftdrink The new release is out, but it takes some time for F-Droid to build it.

saltsoftdrink, 2024-03-10:
> @saltsoftdrink The new release is out, but it takes some time for F-Droid to build it.

Upload in release section of github

wh201906, 2024-03-10:
The apks of older releases are also from F-Droid. I have to wait the build process of F-Droid otherwise the apk signature will change.

closed 2024-03-10 by wh201906

wh201906, 2024-03-14:
The apk can be downloaded on [F-Droid](https://f-droid.org/packages/priv.wh201906.serialtest/) or [GitHub Releases](https://github.com/wh201906/SerialTest/releases/tag/V0.3.4)