wh201906/SerialTest #35: 可以自定义数据协议协议吗？

LandyZou, 2023-06-19:


wh201906, 2023-06-19:
不太明白，可以详细描述一下吗

LandyZou, 2023-06-19:
感谢回复！我想问的是在波形绘制的时候，可以自定义协议解析数据吗？例如一个8字节的数据，数据位只占其中的两个字节，用的是串口协议

----------

该邮件从移动设备发送

-----------------------------------

wh201906, 2023-06-19:
目前还不支持此功能
一个变通的办法是用python进行转发，把串口数据转发到tcp上，在此过程中根据你需要的格式进行转化
方便告知一下你所使用的具体协议格式吗？

LandyZou, 2023-06-19:
一帧数据包括 帧头（4）+有效数据（2） +校验和（1）+附加校验（1）   长度8字节
 
例：
 
AA FF F1 02 08 00 4F
 
详细说明：
 
[0-2]AA FF F1                   固定帧头
 
[3]02                                 有效数据长度0x02（2），STM32 ADC一个通道为u16型数据
 
[4-5]08 00                         ADC采样值，uint16型数据，小端在前 0x0080=128
 
[6]A4                                校验和    前面6字节累加和，取低八位
 
[7]48                                附加校验  前面7字节累加和，取低八位

----------

该邮件从移动设备发送

-----------------------------------

wh201906, 2023-06-19:
前6个字节看懂了，但是第7和第8个你给出的是00 4F，但是解释部分却是A4 48。这一块是写错了吗？
还有一个问题，如果08 00对应0x0080，那么除了大小端以外可能还存在按位逆序或者移位的操作。最好是能多给几组数据包和对应数值的关系方便分析。
另外，用邮件回复格式会乱，方便在github上直接回
复吗？

LandyZou, 2023-06-19:
好的，十分感谢回复，我到github再给出详细的回复

----------

该邮件从移动设备发送

-----------------------------------

LandyZou, 2023-06-19:
一帧数据包括 帧头（4）+有效数据（2） +校验和（1）+附加校验（1）   长度8字节
例：AA FF F1 02 08 00 A4 48
详细说明：
[0-2] AA FF F1			固定帧头
[3] 02				有效数据长度0x02（2），STM32 ADC一个通道为u16型数据
[4-5] 08 00			ADC采样值，uint16型数据，小端在前 0x0080=128
[6] A4 				校验和    前面6字节累加和，取低八位，即：AA + FF +F1 + 02 + 08 + 00 = 2A4（十六进制加法），取低8位为：A4
[7] 48 				附加校验  前面7字节累加和，取低八位

closed 2023-06-19 by LandyZou

wh201906, 2023-06-19:
> [4-5] 08 00 ADC采样值，uint16型数据，小端在前 0x0080=128

这里你确定是0x0080=128而不是0x0008=8吗？后者是标准的小端序，你给出的应该是半字节交换
格式敲定下来之后我可以写个简单的python脚本帮你把收到的串口数据转发到SerialTest上绘图

LandyZou, 2023-06-19:
确定是这样，有个调转

----------

该邮件从移动设备发送

-----------------------------------

wh201906, 2023-06-19:
那如果收到的数据帧第4, 5字节分别是 12 34，对应的采样值是0x4321吗？

LandyZou, 2023-06-19:
是的，十分感谢！

      发自我的企业微信         

 
 
 ----------回复的邮件信息----------

wh201906, 2023-06-19:
大概是这么个样子，因为没设备测试所以你要自己试一下
```
import serial
import socket

# 串口配置
SERIAL_PORT = "COM3"  # 替换为你的串口号
BAUD_RATE = 9600  # 波特率，根据实际情况修改

# TCP服务器配置
SERVER_IP = "0.0.0.0"  # 监听所有可用的网络接口
SERVER_PORT = 12345  # 替换为服务器的端口号

# 数据包格式
PACKET_HEAD = b"\xAA\xFF\xF1"
PACKET_LEN = 8

# 处理数据
def process_data(data):
    ptr = 0
    handled = 0
    values = []
    while True:
        ptr = data.find(PACKET_HEAD, handled)
        if ptr == -1:
            break
        if len(data) - ptr < PACKET_LEN:
            break
        value = data[ptr + 4] + (data[ptr + 5] << 8)
        values.append(value)
        handled = ptr + PACKET_LEN
    return values, handled

# 创建串口对象
ser = serial.Serial(SERIAL_PORT, BAUD_RATE)

# 创建TCP服务器套接字
server_socket = socket.socket(socket.AF_INET, socket.SOCK_STREAM)

try:
    # 绑定服务器地址和端口
    server_socket.bind((SERVER_IP, SERVER_PORT))
    print("服务器已启动，正在监听端口", SERVER_PORT)

    # 开始监听连接
    server_socket.listen(1)

    # 等待客户端连接
    client_socket, client_address = server_socket.accept()
    print("已连接到客户端", client_address)

    values = []
    data = bytearray()
    while True:
        # 从串口读取数据
        new_data = ser.read_all()
        if new_data is not None:
            data += new_data
        new_values, handled = process_data(data)
        for value in new_values:
            client_socket.send(str(value).encode() + "\n")
        data = data[handled:]

except serial.SerialException as e:
    print("串口连接错误:", e)
except socket.error as e:
    print("网络连接错误:", e)
finally:
    # 关闭串口和套接字
    ser.close()
    server_socket.close()

```

LandyZou, 2023-06-19:
好的，非常感谢！我测试一下

      发自我的企业微信         

 
 
 ----------回复的邮件信息----------

LandyZou, 2023-06-19:
请问可以在同一台电脑上运行这个脚本和serialtest tcp服务器吗？

----------

该邮件从移动设备发送

-----------------------------------

LandyZou, 2023-06-20:
虚拟串口，两边都是通的，发送内容也是对的

[image]

但是用你给的那个py脚本做TCP转发，程序运行到读串口的内容的时候，会把最后8位解码成对应的ASCII码，不知道为什么

[image]

[image]

TCP接收到的内容好像也不是很正确

[image]

wh201906, 2023-06-22:
@LandyZou 我测下来是可以正常工作的。你改波特率了吗？
[image]