wh201906/SerialTest #37: 请教：windows下编译经典蓝牙不生效

xiaoyu-ebox, 2023-07-02:
您好，我使用在线安装的qt5.15.2，编译了此工程，发现“蓝牙客户端”、“蓝牙服务器”是置灰的，跟踪代码，发现是auto BTAdapterList = QBluetoothLocalDevice::allDevices(); 这里返回的个数为0，导致经典蓝牙不可用，  但我下载你编译好的，经典蓝牙功能却是正常的。

然后我尝试把下面几句先注释掉，发现经典蓝牙的连接和数据收发也都是正常，只是搜索不到设备（需要提前在电脑系统里面先配对好需要连接的设备）
if(num == 0)
{
    invalid += Connection::BT_Client;
    invalid += Connection::BT_Server;
    invalid += Connection::BLE_Central;
    invalid += Connection::BLE_Peripheral;
}

请问这可能是什么问题呢？你有遇到过吗？是我漏安装了什么吗

wh201906, 2023-07-02:
似乎和 #20 的情况类似。你使用的是MSVC编译器吗？

xiaoyu-ebox, 2023-07-02:
是的，我安装的是MSVC版本，请问你的发行版本是使用什么编译器版本呢？

xiaoyu-ebox, 2023-07-02:
咦，确实和#20的情况一样，那我也换成MinGW版本试下，感谢你的回复

closed 2023-07-02 by xiaoyu-ebox

wh201906, 2024-04-05:
我在 9760cccda87cd8bed31b83e744d615c728153771 当中添加了蓝牙功能对于MSVC编译器的支持，你可以试一下。

目前测下来MSVC编译出来的版本对BLE的支持最好，但是传统蓝牙还是MinGW编译出来的版本要好用些。

wh201906, 2024-04-18:
现在master分支也支持用MSVC编译蓝牙部分的代码了，对应的版本是V0.3.5