wh201906/SerialTest #38: 一点微小的建议

SekiBetu, 2023-07-09:
[image]
发送数据过后能直接清空输入数据的栏的数据吗
然后是，因为有特殊需求，发送的数据不能有换行，导致假如我发送多个数据，下面这个记录发送过的数据的框全都是没有换行的数据

wh201906, 2023-07-09:
看上去你是要给每次发送的数据加换行，那么你需要勾选图片左下角的“后缀”，然后根据需要选择不同的换行符作为结尾。
如果是要一次性发送多行文本，你可以在记事本或者其它文本编辑器当中输入多行文本，然后粘贴到SerialTest当中发送，或者勾选“转义”然后输入\r，\n或者\r\n来达到换行的效果。

wh201906, 2023-07-09:
> 1. 发送数据过后能直接清空输入数据的栏的数据吗

这个功能目前不支持，用到的人似乎也不多。如果你有相关需求可以自行修改代码后编译。这个项目配置了GitHub Action，可以在你fork仓库并修改代码之后自动生成Windows和macOS平台下的可执行文件，不需要自己搭建Qt开发环境

SekiBetu, 2023-07-09:
感谢

SekiBetu, 2023-07-09:
> 看上去你是要给每次发送的数据加换行，那么你需要勾选图片左下角的“后缀”，然后根据需要选择不同的换行符作为结尾。 如果是要一次性发送多行文本，你可以在记事本或者其它文本编辑器当中输入多行文本，然后粘贴到SerialTest当中发送，或者勾选“转义”然后输入\r，\n或者\r\n来达到换行的效果。

我的意思是发送的数据是不带任何换行的，导致显示没有换行，能不能不管用户有没有加换行，都默认给发送日志里的数据换行

closed 2023-07-09 by SekiBetu

SekiBetu, 2023-07-09:
举个例子，我发送 `000000`，然后接着发送 `000001` ，这两个数据都是不带换行符的，所以导致 `已发送` 这个框里面也是没有换行符，挤成一团 `000000000001`

wh201906, 2023-07-09:
我在设计的时候只考虑了原样显示已发送数据的情况，你这种需求也还是要自己改一下代码

https://github.com/wh201906/SerialTest/blob/5f304cfc3df8e74551dc26a15eff1b27392f984f/src/mainwindow.cpp#L514
把括号里面的`data`改成`data+"\n"`应该就好了

SekiBetu, 2023-07-09:
> 我在设计的时候只考虑了原样显示已发送数据的情况，你这种需求也还是要自己改一下代码
> 
> https://github.com/wh201906/SerialTest/blob/5f304cfc3df8e74551dc26a15eff1b27392f984f/src/mainwindow.cpp#L514
> 
> 
> 把括号里面的`data`改成`data+"\n"`应该就好了

好的，感谢解答