wh201906/SerialTest #44: Qt6

bsbhro, 2023-10-28:
When I found this software, my Linux was fully compatible to build it myself. Now I have switched to Ubuntu 23.10. I cannot reconstruct the dependencies to Qt5. Qmake does not give me any errors, but compiling it brings some fatal errors. I, and maybe some other users worldwide, would appreciate if you could make it compatible with Qt6. If it works, I would like to build it on a SBC. Flatpak is not a popular option for me. Thank you!

wh201906, 2023-10-28:
Could you please post the error messages there?

bsbhro, 2023-10-28:
[serTestErrors.txt](https://github.com/wh201906/SerialTest/files/13196060/serTestErrors.txt)
The red color is not adopted :-)

wh201906, 2023-11-03:
> would appreciate if you could make it compatible with Qt6

Sorry. It takes some time to add the Qt6 support so it won't be implemented in the near future.

However, for Ubuntu 23.10, I can build the SerialTest with these commands(I tested them on a live CD environment)
```
sudo apt update
sudo apt install git build-essential
sudo apt install qtbase5-dev qt5-qmake libqt5serialport5-dev qtconnectivity5-dev # I didn't use qt5-default there
# The rest is the same as the steps in README.md
```
Could you please try these steps?

bsbhro, 2023-11-03:
Yes, it linked fine and works. Since I have the program again now, you can take your time. Have a nice weekend!

wh201906, 2023-11-03:
> Have a nice weekend!

Thanks! You too.
I think this issue can be closed now. How do you think?

bsbhro, 2023-11-03:
Sure, Thank you!

closed 2023-11-03 by wh201906