wh201906/SerialTest #45: addition of option "scatter w/o lines"

jojo535275, 2023-11-02:
Hello,

First, thanks a lot for this great tool ! I am using it to monitor an serial output from an FPGA.

Using the serial plotter!
When i send as x axis a sin(x) and as y axis a cos(x) i get the following plot.
[image]

Here i will prefer to see an empty circle ( classic Lissajous figures ) !
Will it be possible to add in the plot tab (next to "scatter") option, a new option of scatter with no lines joining the points of the curve !

Can be named "scatter without lines" , "scatter w/o lines" or "scatter line free" ?

Thanks and have a nice day.

wh201906, 2023-11-02:
That's on my todo list. I need a different graph type in qcustomplot to implement that.

jojo535275, 2023-11-02:
I was think looking at setting the color to fully transparent (alpha=0) in plottab.cpp

So when the "new option" is ticked,  we could execute something like:
currGraph->setPen(QPen(QColor(0, 0, 0, 0)));  // Set the line color to fully transparent

But take your time to do it you way ;-)

wh201906, 2023-11-02:
That's a really good idea! It's much more easier to implement it. Thanks!

wh201906, 2023-11-21:
It's done.
[image]
This feature is in the next release. However, if you want to use it now, you can check the CI builds. The artifacts contain the app and its dependencies.
https://github.com/wh201906/SerialTest/actions/runs/6945872771

jojo535275, 2023-11-21:
hello,

Thanks a lot for this update.
It works great!
If you want, you can have a look at [how the Icestudio community will use your tool to have fun with their fpgas ;-).](https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/O9qP9c8NK-A/m/O6sWY8_VAQAJ)

closed 2023-11-22 by wh201906

wh201906, 2023-11-25:
The latest release is out
https://github.com/wh201906/SerialTest/releases/tag/V0.3.3

jojo535275, 2023-11-25:
Thanks for this one !

have a nice sunday!

commit daf34607 (2023-11-21): Add point-only mode in PlotTab

Fixes #45