wh201906/SerialTest #48: Please introduce more serial file transferring protocol in app like Xmodem, Xmodem CRC, Xmodem 1k, Kermit, Ymodem ,Zmodem, B- plus, Quick Van

Tfjjjastro, 2023-12-01:
Please introduce more serial file transferring protocol in app like Xmodem, Xmodem CRC, Xmodem 1k, Kermit, Ymodem ,Zmodem, B- plus, Quick Van for better for faster and error free file transferring and also android app hangs a lot while file transferring so do something for that its an urgent request, i hope, you may can introduce these features soon
[image]

wh201906, 2023-12-01:
The support for these file transfer protocols is on my todo list, but it takes a lot of time and effort to finish this.

> and also android app hangs a lot while file transferring

Could you please provide more information about this? like what type of communication you used(Serial/TCP/Bluetooth client/...), what device you communicated with and the size of the file.

Tfjjjastro, 2023-12-01:
I have tried to sent 40mb file using Bluetooth clients, it just got hanged after some while, also if possible can you make an extra section in app for voice recording and sending over Bluetooth serial or if it maybe in real time that would be best also this feature should use opus encoding and decoding which can give higher resolution voice at lower bitrate please comment if it is possible and at least try to give one best file transferring option on app from the above list. 
I just can't say, how much this app is helping mine, I really salute to your hardwork, best wishes from my side

wh201906, 2023-12-01:
> I have tried to sent 40mb file using Bluetooth clients, it just got hanged after some while

Yes I just reproduced this bug. I might need to use multi thread to resolve this, but it takes some time to rewrite the related code.

> also if possible can you make an extra section in app for voice recording and sending over Bluetooth serial or if it maybe in real time that would be best also this feature should use opus encoding and decoding which can give higher resolution voice at lower bitrate please comment if it is possible

I think the feature you requested is more suitable for a real-time signal analyzer or relay, which is not what SerialTest is built for. You'd better find a more appropriate app for this.

> and at least try to give one best file transferring option on app from the above list.

I will, but not in the near future.
By the way, I guess you don't need the implementation of a specific protocol because you are seeking for the "one best option". Actually, you don't need to use any protocol if the communication status is good. I also sent a 40MB file in SerialTest from Android to PC, just like you tested before. Although the app does freeze when sending file, but the data sent to the PC doesn't have any corruption(I checked it with CRC32). In all, you could put more trust to the existing communication technology.

wh201906, 2023-12-01:
As a workaround, you can use the `Throttle` feature to prevent the UI freeze.
1. Measure the maximum speed. Just send a file though the Bluetooth. The receiver side is not likely to freeze and you can estimate the speed by checking the `Rx` label at the bottom.
2. Assume that you've got the maximum speed in Step 1, like 50kB/s, then you can set the `Throttle` option in the sender side, like wait 1000ms for every 50000 bytes or wait 100ms for every 5000 bytes.
3. Start sending again. If the actual speed is lower than the maximum speed on the link, the UI freeze won't occur.