wh201906/SerialTest #58: [建议] 指令框能添加特征批量复制功能

akatong, 2024-03-13:
[image]
如图所示，B,C,D,E指令都需要同步添加指令特征点，比如，都需要添加后缀\r\n。
那么目前的做法是手动一个个选中后缀，然后选择\r\n。当指令多起来后，比如100个指令都需要添加后缀，那这样挨个点击，有些不太方便。
看是否可以有个多选选项，开启后，可以将选中的指令批量修改特征。

目前我是先导出一份json表单，然后手动批量修改指令特征的，但这样也不太方便哈。

akatong, 2024-03-13:
[image]

wh201906, 2024-03-13:
这个界面没法做得很复杂，因为要考虑适配手机的问题。如果再加入多选以及其它的按钮会导致手机上用于显示控制项的面积过小

> 目前我是先导出一份json表单，然后手动批量修改指令特征的，但这样也不太方便哈。

目前也只能这样了

akatong, 2024-03-13:
好的，理解，如果是考虑适配其他平台，功能上确实需要精简小巧些。
如果可能的话。或许可以将这个功能仅限于PC端启用，其他平台自适应去diable？或许可以集成。
或许未来可以考虑。🙂

wh201906, 2024-03-13:
好的，后期会考虑加进去

closed 2024-04-06 by wh201906