wh201906/SerialTest #60: [BUG] 发现有种端口软件打开不了也无任何弹窗，其他串口工具可以打开

akatong, 2024-03-13:
[image]
我用sscom可以打开这个端口，正常使用。
但是用SerialTest就不行，有试过将SSCOM和SerialTest全部关掉，然后只开SerialTest,结果还是打不开这个端口，关于这个问题，我如何抓取相关log，协助分析问题呢？

akatong, 2024-03-13:
[image]

wh201906, 2024-03-13:
目前程序里面没做记录log的功能，所以没办法查看。

akatong, 2024-03-13:
[image]

其他端口都不行，这是一台设备，插上USB后，会枚举出以上截图的端口，包括一个调制解调器口。
然后上面的端口都不能连接，SSCOM工具就却可以连接。

wh201906, 2024-03-13:
连接的时候没有任何报错信息吗？按理说连不上是会提示错误的。

akatong, 2024-03-13:
> 连接的时候没有任何报错信息吗？按理说连不上是会提示错误的。

如上图gif所示，我点击了很多次打开按钮，都没有弹出任何提示。

wh201906, 2024-03-13:
FTDI那个端口能打开吗？

akatong, 2024-03-13:
[image]
[image]

FTDI 这个是可以的，这是另一个设备，一个USB转TTL的转接器

wh201906, 2024-03-13:
你能看到设备上使用的移远模块的具体型号吗？

akatong, 2024-03-13:
> 你能看到设备上使用的移远模块的具体型号吗？

EG912UGL
这个型号对分析问题有帮助吗？目前测试下来，SecureCRT，SSCOM 都可以成功连接的🙂

wh201906, 2024-03-13:
本来想买个模块调试一下的，但好像这一款已经买不到了

akatong, 2024-03-13:
> 本来想买个模块调试一下的，但好像这一款已经买不到了

如果我这边通过bus hound 去监听数据交互，这会对分析问题有所帮助吗？
目前点击下打开按钮， 监听到这个数据，但没提示连接成功状态。
[image]

用SSCOM 点击连接，也是监听到这个数据（一样的数据），但是是连接成功，可以发送数据的

wh201906, 2024-03-13:
因为上位机和设备中间隔了一层移远自己的驱动程序，不是走CDC之类通用的协议，因此抓了USB数据也不太好分析

akatong, 2024-03-13:
> 因为上位机和设备中间隔了一层移远自己的驱动程序，不是走CDC之类通用的协议，因此抓了USB数据也不太好分析

如果是这样的话，SSCOM 和SecurtCRT确实做得很好，他们可以兼容并成功连接，SerialTest确实给人很舒服的感觉，是个好开源项目，但遇到这个不兼容问题，感觉有点遗憾 😔，不理解SSCOM这样的老牌串口工具为何兼容性很强。

wh201906, 2024-03-13:
你有试过更新下移远的驱动程序吗？
https://python.quectel.com/download

wh201906, 2024-03-13:
你还可以试一下用CI编译出来的版本
https://github.com/wh201906/SerialTest/actions/runs/8228718937
单击"Artifacts"->SerialTest-MINGW64即可下载
CI编译出来的版本会使用新版本的Qt，可能顺便把这个bug给解决了

akatong, 2024-03-13:
[image]
官网上的是python版的，目前询问了供应商，要了最新版驱动，更新新版驱动后，SerialTest可以成功连接了。
终于解决了问题😁

akatong, 2024-03-13:
哈哈，解决了问题就很开心，SerialTest软件很赞，更多功能还在摸索当中👍

wh201906, 2024-03-17:
我找到连不上的原因了。

SerialTest在成功连接串口之后会尝试设置DTR和RTS，保证UI上的DTR和RTS状态和串口设备一致。然而，移远的老版本驱动不支持这个操作，当程序试图设置这两个信号时会报错。返回的报错信息对应到Qt封装好的串口错误中的QSerialPort::UnknownError，提示内容是“不支持该请求。”。然而按理说这种情况应该返回QSerialPort::UnsupportedOperationError才对。我在处理错误的逻辑当中将QSerialPort::UnknownError视为无法处理或忽略的错误，因此主动关闭了串口。
而移远的新版本驱动虽然也不支持设置DTR和RTS，但是不会返回报错信息，因此能够正常运行。

经过测试，如果在SerialTest当中忽略掉设置DTR和RTS时产生的错误，程序也可以和移远的模块正常通信。这一块我今后会加一个“忽略错误”的开关，用以支持这种非标准情况的错误处理。
[image]

wh201906, 2024-03-17:
> 如果是这样的话，SSCOM 和SecurtCRT确实做得很好，他们可以兼容并成功连接

SerialTest也是能连上的，就是处理错误的时候没考虑过这种不规范的情况，主动断开连接了。

akatong, 2024-03-17:
学习了👍👍👍

commit cae854bc (2024-04-07): Ignore UnknownError when syncing DTR/RTS state

Fixes #60