wh201906/SerialTest #62: 源码重新编译出错

LandyZou, 2024-03-15:
使用的是Qt5.15.2 MinGW 64-bit的环境，直接运行源码报错error: cannot find -lqcustomplotd2

wh201906, 2024-03-15:
你需要下载QCustomPlot的源码，并将qcustomplot.cpp放到src文件夹下，再编译
下载链接可参考README当中的Linux编译步骤

LandyZou, 2024-03-15:
[image]
添加了qcustomplot.cpp之后，报错： error: multiple definition of `QCPVector2D::QCPVector2D()'.....

wh201906, 2024-03-15:
只需要把qcustomplot.cpp放到src/文件夹下，不要修改.pro文件

Lucas-CX, 2024-04-05:
在windows上编译，添加了qcustomplot.cpp，报的和楼主一样的错，请问楼主解决了么？ @LandyZou

LandyZou, 2024-04-05:
@Lucas-CX 我之前下载源码没有添加qcustomplot.cpp，编译出错之后，自己鼓捣了一下，还是没有解决五问题，最后在项目发布者的提示下，添加了qcustomplot.cpp，编译仍然出错，就在想是不是因为之前自己瞎鼓捣修给了一些配置，然后重新解压源码，直接将qcustomplot.cpp添加到src/文件夹下，直接编译就成功编译了。我是使用Qt 5.15.2 MinGW进行编译的。

wh201906, 2024-04-05:
@LandyZou 既然已经编译成功了，那我就先关闭这个issue了
@Lucas-CX 如果你遇到`multiple definition`的错误，请参考[此评论](https://github.com/wh201906/SerialTest/issues/62#issuecomment-1998926287)重新编译。后续有问题可以继续在这个issue下回复

closed 2024-04-05 by wh201906

Lucas-CX, 2024-04-05:
@LandyZou @wh201906 已编译成功，十分感谢各位！！！