wh201906/SerialTest #64: BLE Central not working on Windows

arbrauns, 2024-04-04:
- After connecting to the device using the Windows bluetooth settings screen, it shows up in the Connect tab.
- Clicking "Connect" shows the attribute UUIDs in the right screen, and allows selecting the correct characteristics for RX/TX.
- Upon clicking "Open", the bottom bar changes to show "Status: √" and "Remote: [MAC] Local: 00:00:00:00:00:00", but no data is ever received, and from the device's logs, I can see that the NOTIFY characteristic is never enabled.

Using the latest Windows binary ([V0.3.4-win64.7z](https://github.com/wh201906/SerialTest/releases/download/V0.3.4/V0.3.4-win64.7z)) on Windows 10 Pro 22H2.

wh201906, 2024-04-04:
Would you mind telling me which BLE device you are using?

arbrauns, 2024-04-04:
It's a custom device running the [Zephyr RTOS](https://github.com/zephyrproject-rtos/zephyr).

wh201906, 2024-04-04:
Did you set the Tx characteristic? If so, does it support WRITE operation?

arbrauns, 2024-04-04:
I did set the TX characteristic, but no writes are arriving at the device. Worth noting that the Android app works fine with the same device.

wh201906, 2024-04-04:
Could you please check if this executable is working?
[msvc.zip](https://github.com/wh201906/SerialTest/files/14874902/msvc.zip)

The releases on GitHub are compiled with MinGW, but this one is compiled on MSVC. I think this might help.

wh201906, 2024-04-18:
You can check if the ` V0.3.5-win64-MSVC.7z ` is working for your device.
https://github.com/wh201906/SerialTest/releases/tag/V0.3.5

wh201906, 2025-03-30:
Closed due to inactivity. Feel free to reopen if needed.

closed 2025-03-30 by wh201906