wh201906/SerialTest #75: [求助]控制tab下的滑块功能

KotoriProject, 2024-06-13:
请问这个滑块的数值能否支持16进制？目前是否只有ascii格式？
如果不支持那能否考虑添加这个功能呢
谢谢！

wh201906, 2024-06-13:
滑块目前只支持发送10进制的数字，并且目前感觉用到发16进制数值的人不算多，暂不考虑添加该功能
不过如果你有需求的话可以按照这个patch文件自行修改后再编译。这个项目配置了Windows和macOS的GitHub Action，如果是需要这两个平台上的可执行文件的话你可以fork此仓库，按照patch改好之后在GitHub上运行一次相应的CI，它会自动帮你编译。
[0001-Hex.patch](https://github.com/user-attachments/files/15820251/0001-Hex.patch)

wh201906, 2024-06-13:
根据需求下载Artifacts当中的文件即可
https://github.com/wh201906/SerialTest/actions/runs/9500704927
https://github.com/wh201906/SerialTest/actions/runs/9500704929
https://github.com/wh201906/SerialTest/actions/runs/9500704932

KotoriProject, 2024-06-13:
> 根据需求下载Artifacts当中的文件即可 https://github.com/wh201906/SerialTest/actions/runs/9500704927 https://github.com/wh201906/SerialTest/actions/runs/9500704929 https://github.com/wh201906/SerialTest/actions/runs/9500704932

了解了，感谢回复！

KotoriProject, 2024-06-13:
> 根据需求下载Artifacts当中的文件即可 https://github.com/wh201906/SerialTest/actions/runs/9500704927 https://github.com/wh201906/SerialTest/actions/runs/9500704929 https://github.com/wh201906/SerialTest/actions/runs/9500704932

我刚刚分别下载了您修改好的windows d msvc64和MINGW64两个版本进行了尝试，我想您可能误解了我的意思，我的意思是数值从0~255（即0x00~0xFF），而串口也以此源数据发送。
我现在尝试后得到的结论是：您将数值变化从十进制修改成了16进制，但串口发送时仍然是将数值转换为ascii格式再发送，那么这种发送方式我认为也很少用到；
但按照我的原意应当是：数值变化是正常是十进制，但是串口发送时，直接发送原数值，例如：数值0→发送00（16进制显示下），数值255→发送FF（16进制显示下）。我想这种应用场景还是很多的，您在控制模组的设置中也允许前后缀使用“16进制”格式，何不让变化的数值也支持使用“16进制”格式呢？
我是一名嵌入式初学者，对于软件这方面不是很熟悉，说了这么多而我无法帮忙提pr，还请作者海涵，谢谢。

wh201906, 2024-06-16:
我之前误解你的意思了。大概今天晚上的样子我会改好发给你

KotoriProject, 2024-06-16:
谢谢😇❤️

wh201906, 2024-06-17:
这是改好的版本
https://github.com/wh201906/SerialTest/actions/runs/9546935880
https://github.com/wh201906/SerialTest/actions/runs/9546935867
https://github.com/wh201906/SerialTest/actions/runs/9546935866

> 数值变化是正常是十进制，但是串口发送时，直接发送原数值，例如：数值0→发送00（16进制显示下），数值255→发送FF（16进制显示下）。我想这种应用场景还是很多的

确实，直接以16进制发送的话接收端就不需要对字符串进行转换了，我看看后期有空能不能加一下这个功能

KotoriProject, 2024-06-17:
谢谢大佬耐心修改！没有问题了

closed 2024-06-17 by KotoriProject